deltaphi/c6021light #14: Only F0-F8 forwarded between CAN and LocoNet

glaserf, 2021-02-24:
The LocoNet personal edition only specifies commands up to F8. It seems LN is capable of higher functions as well. The following encoding was reverse-engineered by snooping on the LN-T port of an Intellibox II:
```
0xA3 slot_id 0b0000_[F12:F9 in Bits3:0]

0xD4 0x20 slot_id  0x8 0b0[F19:F13 in Bits 6:0]
0xD4 0x20 slot_id  0x9 0b0[F27:F21 in Bits 6:0]
0xD4 0x20 slot_id  0x5 0b0[F28|F20 in Bits 6:5]0_0000
```

glaserf, 2021-02-24:
The above was deducted from the following raw output:
```
LN RX:  a3  1  1 5c -- F9  ON
LN RX:  a3  1  0 5d -- F9  OFF
LN RX:  a3  1  2 5f -- F10 ON
LN RX:  a3  1  0 5d -- F10 OFF
LN RX:  a3  1  4 59 -- F11 ON
LN RX:  a3  1  0 5d -- F11 OFF
LN RX:  a3  1  8 55 -- F12 ON
LN RX:  a3  1  0 5d -- F12 OFF

LN RX:  d4 20  1  8  1  3 -- F13 ON
LN RX:  d4 20  1  8  0  2 -- F13 OFF
LN RX:  d4 20  1  8  2  0 -- F14 ON
LN RX:  d4 20  1  8  0  2 -- F14 OFF
LN RX:  d4 20  1  8  4  6 -- F15 ON
LN RX:  d4 20  1  8  0  2 -- F15 OFF
LN RX:  d4 20  1  8  8  a -- F16 ON
LN RX:  d4 20  1  8  0  2 -- F16 OFF
LN RX:  d4 20  1  8 10 12 -- F17 ON
LN RX:  d4 20  1  8  0  2 -- F17 OFF
LN RX:  d4 20  1  8 20 22 -- F18 ON
LN RX:  d4 20  1  8  0  2 -- F18 OFF
LN RX:  d4 20  1  8 40 42 -- F19 ON
LN RX:  d4 20  1  8  0  2 -- F19 OFF

LN RX:  d4 20  1  9  1  2 -- F21 ON
LN RX:  d4 20  1  9  0  3 -- F21 OFF
LN RX:  d4 20  1  9  2  1 -- F22 ON
LN RX:  d4 20  1  9  0  3 -- F22 OFF
LN RX:  d4 20  1  9  4  7 -- F23 ON
LN RX:  d4 20  1  9  0  3 -- F23 OFF
LN RX:  d4 20  1  9  8  b -- F24 ON
LN RX:  d4 20  1  9  0  3 -- F24 OFF
LN RX:  d4 20  1  9 10 13 -- F25 ON
LN RX:  d4 20  1  9  0  3 -- F25 OFF
LN RX:  d4 20  1  9 20 23 -- F26 ON
LN RX:  d4 20  1  9  0  3 -- F26 OFF
LN RX:  d4 20  1  9 40 43 -- F27 ON
LN RX:  d4 20  1  9  0  3 -- F27 OFF

LN RX:  d4 20  1  5 20 2f -- F20 ON
LN RX:  d4 20  1  5  0  f -- F20 OFF
LN RX:  d4 20  1  5 40 4f -- F28 ON
LN RX:  d4 20  1  5  0  f -- F28 OFF
```

deltaphi, 2021-02-27:
Unit testing infrastructure for this issue is available as of fae21ef3e557b923aff3c2df2aa8ebc513e9918f.

commit e210a526 (2021-02-28): Add support for F20..28

Fixes #14.
closed 2021-02-28 by deltaphi